adafruit/Adafruit_GPS #40: Checksum code nonfunctional

nmcross, 2015-04-21:
Your checksum code is actually inactive.  If you look at the checksum portion of the parse() command you can see that the "return false" when the checksum is incorrect is commented out.  That's because the code doesn't currently work, at the conclusion of every checksum check sum=36 which happens to be the value of nmea[1].  If you instead run the for loop from i=2 to strlen(nmea)-4, the code actually functions as anticipated.

ladyada, 2015-05-18:
ya - addressed here (with a fix)
http://forums.adafruit.com/viewtopic.php?f=22&t=73792
will check out!

PaulZC, 2015-11-08:
I think the root cause of this is the last few lines in the GPS::read function. They add the \n to the start of the new line, which in turn moves the $ across into the checksum calculation. If they were wrapped in an else statement, I think the problem would go away...:

  // I think this bit needs to be in an else statement otherwise the \n gets added
  // to the start of the new line and messes up the checksum check
  else { 
    currentline[lineidx++] = c;
    if (lineidx >= MAXLINELENGTH)
      lineidx = MAXLINELENGTH-1;
  }

closed 2015-11-30 by ladyada