adafruit/Adafruit_GPS #41: Bad Time Data Every 4 Seconds

umop-apisdn, 2015-06-04:
It seems the Adafruit-GPS is returning bad time data every 4 seconds because the time is being parsed internally as a floating point value.  This causes a round-off error that causes the time with seconds of 1, 5, 9, 13, 17, 21, 25, 29, 33, 37, 41, 45, 49, 53 and 57 to be repeated.  This problem happens between about 09:55 and 24:00 GMT.  I've asked about this problem on the Adafruit forum, but figured I'd ask here since the issue seems to be in the library.  It looks to me if the time in the RMC and GGA sentences was parsed using a series of 4 atoi commands instead of one atof command, this problem would be solved.

antoniosap, 2015-06-24:
please write down code update

umop-apisdn, 2015-06-24:
Antonio,

 

In the Adafruit_GPS library the problem I had was with the “parse” routine.  The cpp code was treating the time from the GPS as a floating point number that was being rounded off in a strange way.  It was using the atof function.

 

My solution was to convert the parts of the time as integers using the atoi function.  I added the variable “char timebuff[10]”.

 

I needed to change code in two places.

1.       The section that parses the GPGGA data

2.       The section that parses the GPRMC data

 

Here is the modified code that is used in the GPGGA section:

 

  int32_t degree;

  long minutes;

  char degreebuff[10];

  char timebuff[10];   // << Added

 

  // look for a few common sentences

  if (strstr(nmea, "$GPGGA")) { 

    // found GGA

    char *p = nmea;

    // get time

       p = strchr(p, ',')+1;

timebuff[2] = '\0';

 

       strncpy(timebuff, p, 2 );

       hour = atoi(timebuff);

 

       strncpy(timebuff, p+2, 2 );

       minute = atoi(timebuff);

 

       strncpy(timebuff, p+4, 2 );

       seconds = atoi(timebuff);

 

       timebuff[3] = '\0';

       strncpy(timebuff, p+7, 3 );

       milliseconds = atoi(timebuff);

 

Here is the modified code in the GPRMC section:

 

  if (strstr(nmea, "$GPRMC")) { 

   // found RMC

    char *p = nmea;

 

    // get time

    p = strchr(p, ',')+1;

       timebuff[2] = '\0';

 

       strncpy(timebuff, p, 2 );

       hour = atoi(timebuff);

 

       strncpy(timebuff, p+2, 2 );

       minute = atoi(timebuff);

 

       strncpy(timebuff, p+4, 2 );

       seconds = atoi(timebuff);

 

       timebuff[3] = '\0';

       strncpy(timebuff, p+7, 3 );

       milliseconds = atoi(timebuff);

 

I hope this helps.  I have not worked with github before and don’t know the procedures to submit code.  If you know how to submit changes, please feel free to submit these changes.  I’d like if you include my name in the submission.  My name is John Nelson.

 

Thank you,

John

antoniosap, 2015-06-25:
this solve the .984 on milliseconds.
thanks

tonton81, 2016-04-14:
data randomly changing every few seconds is also a baud rate issue, i decreased from 115200 10hz to 57600 @ 5hz and no more problems.

HamishB, 2018-10-03:
Hi, here is a patch to solve the milliseconds value being corrupted due to lack of space in a single precision floating point to store hhmmss.iii

```
--- Adafruit_GPS.cpp.ORIG       2018-10-03 21:43:26.022428754 +1300
+++ Adafruit_GPS.cpp    2018-10-03 23:02:51.023279702 +1300
@@ -62,7 +62,9 @@
     minute = (time % 10000) / 100;
     seconds = (time % 100);
 
-    milliseconds = fmod(timef, 1.0) * 1000;
+    // hhmmss.iii won't fit in float so we parse a bit more
+    p = strchr(p, '.')+1;
+    milliseconds = atoi(p);
 
     // parse out latitude
     p = strchr(p, ',')+1;
@@ -166,7 +168,8 @@
     minute = (time % 10000) / 100;
     seconds = (time % 100);
 
-    milliseconds = fmod(timef, 1.0) * 1000;
+    p = strchr(p, '.')+1;
+    milliseconds = atoi(p);
 
     p = strchr(p, ',')+1;
     // Serial.println(p);
```

For millisec values less than .100 you will still need to make a nice way to print them in your code, as C's printf("%03d") in C++ involves the less elegant #include <iomanip>'s setfill() and setw(). Fortunately(???) the Adafruit Ultimate GPS shield maxes out ato 10Hz so at least with that unit you won't see anything between .000 and .100 sec.

regards,
Hamish

HamishB, 2018-10-04:
In doing some 1 minute logging overnight I came across something which means the above patch might need a bit more adjustment.

The Adafruit Ultimate GPS board set to output GGA and RMC at 5Hz produced an NMEA string with a funny timestamp: `$GPGGA,142159.999,`. The atof() call rounded this to 142160 and then adding the .999 part on separately ended up with output like 60.999. Possible solution is to do away with the timef variable and use atoi() on the hhmmss part of the string, assuming that stops at the '.'.

Hamish

umop-apisdn, 2018-10-04:
Hamish,

 

It’s been a long time since I reported the problem I was having but to the best of my recall the problem showed up mostly when the time of day passed about 20:00 UTC.  I think it had something to do with exceeding the ability of the float type variable to handle the full timestamp.

I’ll have to go back and check but I think as a work around I parsed the string from the GPS module in 2 parts instead of one or maybe parsed it as multiple integers.  I rolled my changes into my own GPS library with most of the Adafruit code and used it.  

With updates to the Arduino IDE and library updates my mods may not even be on my system any more.

I think you’re right about using 5Hz or 10Hz output rates having problems too.

I’ll see what I can find.

 

Thank you for looking into this,

John Nelson

closed 2019-06-17 by drak7