adafruit/Adafruit_GPS #43: idDigit() and isAlpha() undefined

phieber, 2015-08-05:
the two functions are undefined 

/home/phieber/Downloads/arduino-1.6.5/libraries/Adafruit-GPS-Library/Adafruit_GPS.cpp: In member function 'boolean Adafruit_GPS::LOCUS_ReadStatus()':
/home/phieber/Downloads/arduino-1.6.5/libraries/Adafruit-GPS-Library/Adafruit_GPS.cpp:454:20: error: 'isDigit' was not declared in this scope
if (isDigit(c))
^
/home/phieber/Downloads/arduino-1.6.5/libraries/Adafruit-GPS-Library/Adafruit_GPS.cpp:463:24: error: 'isAlpha' was not declared in this scope
if (isAlpha(parsed[2])) {
^
Error compiling

tdicola, 2015-08-05:
What platform/board are you using?  It looks like it might be an issue with that board not implementing the same functions as the normal Arduino cores like the Uno, Due, etc.

phieber, 2015-08-05:
I use the TeensyLC

microbuilder, 2015-12-28:
Sorry for the slow reply ... I just ran into this myself on a different ARM board.

This is due to a difference in the standard C++ library function names.  Most ARM toolchains include a libc with lower case isdigit, isalpha (etc.) functions in ctype.h, but the Arduino libraries that most drivers are written against use isDigit, isAlpha, etc., with an upper case character.

It's a hack, but as a quick fix you can open up the main .c file and insert an #ifdef like this to toggle between the two, you just need to find an appropriate test value in the define for your platform (Teensy, etc.):

```
      #if defined(ARDUINO_ARCH_STM32F2)
      if (isdigit(c))
      #else
      if (isDigit(c))
      #endif
```
And the same for isAlpha:
```
  #if defined(ARDUINO_ARCH_STM32F2)
  if (isalpha(parsed[2])) {
  #else
  if (isAlpha(parsed[2])) {
  #endif
```

closed 2015-12-28 by microbuilder