adafruit/Adafruit_GPS #52: Missing zeros in GPS.lat // GPS.lon

Verm1ll1on, 2015-11-07:
Hi there, I'm really new to arduinos (or in general with coding, started 3 days ago ) and have to use the adafruit gps for a university project  when i'm using the standard method GPS.lat or GPS.lon 
the method will give me 
Location : 5345.2211N, 940.3187E
from this GGA line
$GPGGA,110758.000,5345.2215,N,00940.3184,E,1,05,3.42,-7.2,M,45.8,M,,*44

is there a possibility to rewrite the method so that one Zero will not be deleted so that you could use those coordinates directly in google maps ? 

Sorry for such a newbie question ;) 

Thanks for you help

ValentmSTEM, 2015-11-25:
Four decimal places will give you 11m accuracy on Google Maps.

tonton81, 2016-04-14:
just change the ",4" from serial.print of longitude/latitude to ",6" for example will give you 6 decimal places..

closed 2019-05-16 by drak7