adafruit/Adafruit_GPS #56: Arduino 101 Compatibility

Arvind-TJHSST, 2016-02-26:
This library seems to be incompatible with the Arduino 101, with the line Adafruit_GPS GPS(&mySerial); in the example sketch giving the error "exit status 1
no matching function for call to ‘Adafruit_GPS::Adafruit_GPS(SoftwareSerial*)’"
I would appreciate some help to get this modified to work with the 101. Thanks.

josephmeehan, 2016-04-18:
I get the same error, I have the SD card working with the Adafruit GPS logger and Arduino 101 but not the GPS, same SoftwareSerial error.  Is it possible to get a modification?  Or tips on how to make the modifications?

josephmeehan, 2016-04-29:
Use HW Serial instead of SW serial and move the #endif in the begin function in Adafruit_GPS.cpp file above the HW serial begin call.  It works for me now.

ladyada, 2016-07-12:
w00pz - fixed this and also added HwSerial demos for both Echo & Parsing!

closed 2016-07-12 by ladyada

orkgks, 2017-03-13:
Arduino/Genuino MKR1000 doesn t work with adafruit gps library.Can you resolve the error or write a new library for arduino / genuino

robotdna, 2019-01-25:
I am using HW serial instead of SW serial, my .cpp file has the HW serial call after the SW serial endif. This is not working for me- using Arduino 101.