adafruit/Adafruit_GPS #58: Adafruit_GPS on Adalogger Feather M0

PaulZC, 2016-04-09:
I've been having fun connecting an Ultimate GPS Breakout (V3) to an Adalogger M0 (hook-up details are appended below):

There are no examples for the Feather (yet) so I used code for the Flora:

The flora_gpstest example works straight out of the box. 

To get flora_parsing to work, you need to make a few changes:

In flora_parsing:
Comment out the include for SoftwareSerial.h (line 14):
```Arduino
//#include <SoftwareSerial.h>
```

In the library Adafruit_GPS.cpp:
In the code for Adafruit_GPS::begin:
Add three lines just above the delay(10); (line 362):
```Arduino
void Adafruit_GPS::begin(uint16_t baud)
{
#ifdef __AVR__
  if(gpsSwSerial) 
    gpsSwSerial->begin(baud);
  else 
    gpsHwSerial->begin(baud);
#endif

#ifdef SAM // For the Adalogger Feather M0
  gpsHwSerial->begin(baud);
#endif

  delay(10);
}
```

After that, everything seems to work great!

(I haven't pulled this as it raises much bigger questions about softwareserial on the M0 in general. I'll leave it to Ada and the team to figure out the best approach for this!)

Hook-Up
=======

|Ultimate GPS|Adalogger M0|
|----------|----------|
|TX|RX0 (D0)|
|RX|TX0 (D1)|
|GND|GND|
|VIN|USB (VBUS)|

The Ultimate GPS draws power from the USB port or the LiPo battery and doesn't put any load on the Adalogger 3.3V regulator. Connecting VIN this way is OK as the Ultimate GPS has its own built-in regulator. You will need to choose a different power pin if your GPS module doesn't have a regulator!

PaulZC, 2016-04-09:
This is the same issue (and almost the same fix) as Pull #55

ladyada, 2016-07-12:
w00pz - fixed this and also added HwSerial demos for both Echo & Parsing!

closed 2016-07-12 by ladyada

mimiflynn, 2016-11-28:
I believe I'm running into the same or similar issue when attempting to follow [this tutorial](https://learn.adafruit.com/adafruit-ultimate-gps-featherwing/arduino-library) with a Feather M0.

I'm using `flora_gpslog.ino` instead of `flora_parsing.ino` per the tutorial since the latter is no longer available in the current master branch that I'm running.

```
/Documents/Arduino/libraries/Adafruit_GPS/examples/flora_gpslog/flora_gpslog.ino:2:28: fatal error: SoftwareSerial.h: No such file or directory
 #include <SoftwareSerial.h>
                            ^
compilation terminated.
exit status 1
Error compiling for board Adafruit Feather M0 (Native USB Port).
```

Please advise.

ladyada, 2016-11-28:
use the hardware serial demo!