adafruit/Adafruit_GPS #67: If checksum is corrupt don't assume message is OK

snorfalorpagus, 2017-08-07:
If an NMEA checksum is missing the library will assume everything is OK and continue to parse the message. This can cause problems if the message is so corrupt that the checksum itself is also corrupted, i.e. the `*` is missing or in the wrong place. I think a better approach would be to skip the message if the checksum is missing. The GPS unit on the v3 board always seems to have a checksum, so I can't see a benefit to ignoring missing values.

Relevant code here: https://github.com/adafruit/Adafruit_GPS/blob/master/Adafruit_GPS.cpp#L36

SamuelWAnderson45, 2017-11-16:
I've had to manually patch this for myself to prevent collecting bad data from the GPS. It would be good to see this fixed on master

billpugh, 2018-09-10:
I agree. I encountered several situations in which the strings coming from the GPS unit were corrupted such that there wasn't a * in the expected column and thus the checksum check was ignored.

snorfalorpagus, 2018-09-11:
This repository hasn't seen a commit for a year and I think it's fair to say it's stagnant. There are plenty of other NMEA parsers available for Arduino. I have had some luck with TinyGPS.

drak7, 2019-04-10:
Good idea, adding this to the library.

closed 2019-04-10 by drak7