adafruit/Adafruit_GPS #68: Millisecond output bug in several parsing examples

JRDavisUF, 2017-10-04:
In the examples parsing and due_parsing (and perhaps others), time is output using code which looks like the following:

Serial.print("\nTime: ");
Serial.print(GPS.hour, DEC); Serial.print(':');
Serial.print(GPS.minute, DEC); Serial.print(':');
Serial.print(GPS.seconds, DEC); Serial.print('.');
Serial.println(GPS.milliseconds);

Because GPS.milliseconds is actual milliseconds:
     (unit16_t) milliseconds = fmod(timef, 1.0) * 1000;
the millisecond value needs to have some leading zeros added.  Two if: value < 10 and one if:  9 < value < 100.

For example, say the seconds value of time is 3  seconds and 9 milliseconds, the value output by the code above would be 3.9, when it should be 3.009.

HamishB, 2018-10-03:
either the lib needs to store millisec as a string (new GPS.millistr variable?) or you need to recreate C's printf("%03d") in C++.  Study the web for #include <iomanip>'s setfill() and setw() and post the example code if you feel kind, it could make its way into one of the library's examples.

see also my millisec patch in bug #41, and see also bug #52.

regards,
Hamish

HamishB, 2018-10-04:
try this:
```
if(mils < 10)
  Serial.print("0");
if(mils < 100)
  Serial.print("0");
Serial.print(mils);
```

Hamish

closed 2019-06-17 by drak7