adafruit/Adafruit_GPS #75: Getting fix false all the time

StepanyanAlbert, 2018-07-09:
Arduino IDE 1.8.5 . Using example 
Time: 0:0:0.0 AdafruitGPSLibrary -> parsing 
left over 1 hour still not getting *FIX* to true 
Even though 
`if (strstr(nmea, "$GPRMC")) {
   // found RMC
    char *p = nmea;

    // get time
    p = strchr(p, ',')+1;
    float timef = atof(p);
    uint32_t time = timef;
    hour = time / 10000;
    minute = (time % 10000) / 100;
    seconds = (time % 100);

    milliseconds = fmod(timef, 1.0) * 1000;

    p = strchr(p, ',')+1;
    // Serial.println(p);
    if (p[0] == 'A') 
      fix = true;
    else if (p[0] == 'V')
      fix = false;
    else
      return false;`

so if there is an A fix should be true and you can see from serial output  A instead of V
`
Date: 0/0/200
Fix: 0 quality: 0
$GPRMC,055009.00,A,4012.35535,N,04429.86519,E,0.495,,090718,,,A*7C
$GPVTG,,T,,M,0.495,N,0.917,K,A*24
$GPGGA,055009.00,4012.35535,N,04429.86519,E,1,05,2.08,1158.8,M,8.6,M,,*5B
$GPGSA,A,3,19,17,24,15,12,,,,,,,,3.88,2.08,3.27*02
$GPGSV,3,1,09,02,19,169,,06,38,126,21,12,35,288,43,13,17,196,*77
$GPGSV,3,2,09,15,25,23423,1,47,47,2619,74058,1,2459,31,40*D
$PGSV3,3,0,28,1,081,1*41
$GPGL,402.3535,N,4429.6519E,05009.0,A,*6A
$GPRMC,055010.00,A,4012.35513,N,04429.86543,E,0.177,,090718,,,A*76
$GPVTG,,T,,M,0.177,N,0.327,K,A*24
$GPGGA,055010.00,4012.35513,N,04429.86543,E,1,05,2.08,1158.4,M,8.6,M,,*54
$GPGSA,A,3,19,17,24,15,12,,,,,,,,3.88,2.08,3.27*02
$GPGSV,3,1,09,02,19,169,,06,38,126,20,12,35,288,43,13,17,196,*76
$GPGSV,3,2,09,15,25,34,2,17,4,047,7,19,4,05821,2459,31,407B
$PGSV3,3,9,28,3,08112*4
$GGLL,012.5513N,0429.8643,E,5501000,AA*69
$GPRMC,055011.00,A,4012.35440,N,04429.86566,E,0.400,,090718,,,A*72
$GPVTG,,T,,M,0.400,N,0.742,K,A*26
$GPGGA,055011.00,4012.35440,N,04429.86566,E,1,05,2.08,1158.2,M,8.6,M,,*53
$GPGSA,A,3,19,17,24,15,12,,,,,,,,3.88,2.08,3.27*02
$GPGSV,3,1,09,02,19,169,,06,38,126,21,12,35,288,43,13,17,196,*77
$GPGSV,3,2,09,15,25,234,2,17,4,047,8,19,4,05821,2459,30,40*7
$GGSV,,3,0928,13081,1*44
GPGL,401.3540,N,0429.6566E,05011.0,A,A68
`

francose, 2018-07-10:
I have the same problem, any solution  ??

StepanyanAlbert, 2018-07-10:
are devs still supporting project or we need to use diff lib ?
I was thinking problem is the sensor but it works with plain Serial

francose, 2018-07-11:
I actually changed the bund rate to 4800  and it started to populate the data.

StepanyanAlbert, 2018-07-11:
@francose  where exactly?

Dygear, 2019-06-25:
@francose what was your baud rate before? I'm wondering if it's too fast or too slow before. I'd also like to know on what platform you're using it on. I've had zero issues like this using the Hardware Serial interface, but YMMV.

francose, 2019-06-26:
@Dygear it was 115000band rate. Apparently it was a device issue that we were using. The data we were getting wasnt accurate so we ended up write our routine for that specific device. (GPS --> Byionics )

Dygear, 2019-06-27:
@francose So you're saying this issued was resolved? Can you close this issue if that is the case?

francose, 2019-06-27:
@Dygear i didnt open this issue neither i m saying that this issue has been resolved. I just participated above comments. I thought my issue was relative to this opened issue which wasnt really..

ladyada, 2019-07-09:
thanks, closin!

closed 2019-07-09 by ladyada