adafruit/Adafruit_GPS #76: Missing documentation about how to decode `latitude_fixed` and `longitude_fixed`

MichaelMauderer, 2018-12-04:
The library includes a high precision integer encoded value for latitude and longitude in the form of  `latitude_fixed` and `longitude_fixed`. But there is no documentation or example on how to decode those values. 

The forum contains this example using string manipulation (probably not ideal and does produce wrong results for me).
https://forums.adafruit.com/viewtopic.php?f=57&t=110423&p=551739&hilit=longitude_fixed#p551856

Dygear, 2019-06-25:
What exactly are you looking for here? In line comments in the code, or something else?

MichaelMauderer, 2019-07-26:
It's been a while since I looked at this. If I remember correctly, the ideal solution would be to have a helper function that does the conversion. Some additional inline comments might be enough, too, though.

sellensr, 2020-01-31:
int32_t latitude_fixed;  ///< Fixed point latitude in decimal degrees.
                           ///< Divide by 10000000.0 to get a double.
  int32_t longitude_fixed; ///< Fixed point longitude in decimal degrees
                           ///< Divide by 10000000.0 to get a double.
will appear soon in the source code. There is no encoding, just a factor of 10^7.

closed 2020-02-10 by drak7