adafruit/Adafruit_GPS #95: NMEA TXT message

hydrastarmaster, 2019-09-11:
Please implement the NMEA TXT sentence parsing.
Official document here: https://www.nmea.org/Assets/20160520%20txt%20amendment.pdf

Thank you.

ladyada, 2019-09-11:
sounds cool - we dont have any devices that emit this sentence - if you have one, please try it out and submit a PR!

hydrastarmaster, 2019-09-11:
Never seen it. NEO 6M specs indicate it will produce this sentence. It is a very common and cheap module. It seems to be an on-demand message, so, not easy to catch in the every-day transmissions.
I've noticed that it's the only sentence that has never been implemented in libraries.

ladyada, 2019-09-11:
we don't own any of those modules, if someone has one and submits a PR we'll check it out!

hydrastarmaster, 2019-09-11:
Collected several examples from different UBLOX devices:

```
$GPTXT,01,01,25,DR MODE - ANTENNA FAULT^21*38

$GPTXT,01,01,01,ANTENNA OK*35

$GNTXT,01,01,00,txbuf alloc*61
$GPTXT,01,01,01,NMEA unknown msg*58

$GPTXT,01,01,00,txbuf alloc*7F

$GNTXT,01,01,02,u-blox AG – http://www.u-blox.com*4E
$GNTXT,01,01,02,HW UBX-M8030 00080000*60
$GNTXT,01,01,02,ROM CORE 3.01 (107888)*2B
$GNTXT,01,01,02,FWVER=SPG 3.01*46
$GNTXT,01,01,02,PROTVER=18.00*11
$GNTXT,01,01,02,GPS;GLO;GAL;BDS*77
$GNTXT,01,01,02,SBAS;IMES;QZSS*49
$GNTXT,01,01,02,GNSS OTP=GPS;GLO*37
$GNTXT,01,01,02,LLC=FFFFFFFF-FFFFFFFF-FFFFFFFF-FFFFFFFF-FFFFFFFD*2F
$GNTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*3E
$GNTXT,01,01,02,ANTSTATUS=OK*25
$GNTXT,01,01,02,PF=3FF*4B

$GPTXT,01,01,02,u-blox ag - www.u-blox.com*50
$GPTXT,01,01,02,HW  UBX-G60xx  00040007 FF7FFFFFp*53
$GPTXT,01,01,02,ROM CORE 7.03 (45969) Mar 17 2011 16:18:34*59
$GPTXT,01,01,02,ANTSUPERV=AC SD PDoS SR*20
$GPTXT,01,01,02,ANTSTATUS=DONTKNOW*33
```
Note the use of "^" to introduce special chars.

jsleeio, 2019-09-12:
@hydrastarmaster it seems that you already have the required info for submitting a PR?

closed 2020-02-10 by drak7