adafruit/Adafruit_GPS #99: Parsing of the record (for issue GPGGA) with empty values brings not actual values for HDOP and some other positions.

irapu, 2019-10-07:
- Arduino board:  **Genuino/UNO**

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.9

i'm requesting the record only for GGA :
`GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_GGAONLY); `

if i receive complete record like this:
`$GPGGA,185526.000,4810.5543,N,01139.0849,E,1,05,1.52,488.0,M,47.5,M,,*65`
everything works fine and i get correct values
```
GPS.satellites = 5;
GPS.HDOP = 1.52;
```
But if after that commes a record wih empty values like 
`$GPGGA,184426.000,,,,,0,04,,,M,,M,,*71`
i will get the last read value for **HDOP**. It is not correct for me:
`GPS.satellites = 4;
GPS.HDOP = 1.52;`

i'm not a c++ developer, but i think, the reason is on this point:
from line 121 in Adafruit_GPS.cpp:
```
  p = strchr(p, ',')+1;
    if (',' != *p)
    {
      HDOP = atof(p);
    }
// from hear is my code:
	else
	{
		HDOP = 0.00;
	}	

```

could you please take a look at this case?
i have a workaround, i parse a record myself to take a HDOP at it's position.
But it would be better to have suitable version from original ))

Thank you in advance

sellensr, 2019-10-08:
IIRC the library intentionally retains the most up to date information, allowing you to access the last fix until it is updated. You could use secondsSinceFix() to assess whether the information is up to date. Also, you probably don’t want to pretend HDOP is zero. A better estimate of HDOP might be SOG * secondsSinceFix() with appropriate unit conversion.

Cheers,

Rick

irapu, 2019-10-09:
Thank you, I will try your solution with secondsSinceFix().

closed 2019-10-09 by irapu