adafruit/Adafruit_GPS #102: Data quality issue, simple fix?

basrijn, 2019-11-07:
Hi,

I noticed that my Adafruit Altimate GPS wing (and I would assume any MTK3339 based board) is reporting a satellite count, even if no fix is obtained:

`$GPGGA,063633.000,,,,,**0**,**08**,,,M,,M,,*73`

This results in GPS.satellites containing a value where it really should read zero if there is no fix obtained.

Updating the GPGGA parser to read:
```
satellites = atoi(p);
      if (!fix)
        satellites = 0;
```
Would address that data quality issue.

Or is there a known reason for the chip to actually report this value. GGA is fix related data, so it would make for the satellite field to reflect fix satellites (as it does after fix is obtained). The GSV sentence has a satellite field that maps to satellites in view if somebody would ever want that data.

 Bas

sellensr, 2019-11-07:
A GPS unit can have contact with more than 0 satellites while still being unable to resolve a reliable fix. I think it would be unwise to eliminate information provided by the unit when parsing the sentence.

Cheers,

Rick

basrijn, 2019-11-07:
Hi Rick,

Unfortunately I have not been able to find a free copy of the NMEA 0183 standard. I always understood the GGA sentence to be fix data, and the GSV sentence to be Sat in View data. But I found at least [one document](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:advanced_features:nmea_sentences) where they describe GGA also as Sats in view.

I will close this and add a parser for the GSV sentence to see if the number reported in GGA and GSV is the same. If not, I will re-open this.

Bas

closed 2019-11-07 by basrijn

caternuson, 2019-11-07:
"satellite count" is simply the number of satellites currently being listened to. If the signal is weak, it can take some time to obtain a good "fix". This isn't really specific to any data sentence format. It's just a fundamental aspect of how GPS works.

This is essentially the same thing @sellensr is saying.

I think "satellites in view" can also be confusing. That can come from the almanac of known satellites, current time, and last known good location. It will get updated once good communication is established and actual location becomes known.

"satellites in view" = total number of satellites overhead
"satellite count" = actual number of satellites that are currently being talked to
"fix" = a position has been computed, requires multiple good satellite signals