adafruit/Adafruit_GPS #114: parseLatDir and parseLonDir difference

loky32, 2020-01-29:
Hi, I write because found a difference between this two functions that need similar treatment:

https://github.com/adafruit/Adafruit_GPS/blob/24715012b3dcf05e47073e523d2ad5aac4ad2ed5/src/Adafruit_GPS.cpp#L233-L246

https://github.com/adafruit/Adafruit_GPS/blob/24715012b3dcf05e47073e523d2ad5aac4ad2ed5/src/Adafruit_GPS.cpp#L282-L297

My question is, ¿Why in latitude the condition (if (!isEmpty(p))) is not requiered?
I think that this is a problem because when the value is not present, if you are in the south of Ecuador your latitude value can change of sign.

sellensr, 2020-01-29:
Good catch! Coordinates always arrive as DDMM.mmmm,S for a southern latitude, so it should always find an S and does in my tests. In any case I hope it will soon be moot, as parseCoord() will replace all four lat lon in the pending pull request.

sellensr, 2020-01-29:
The new parseCoord() function explicitly tests that the direction field is not empty, that the direction field is one of N, E, S, W, and that the magnitude of the angle agrees with the direction. https://github.com/adafruit/Adafruit_GPS/pull/113 was merged about an hour ago, so the issue is solved.

closed 2020-02-10 by drak7