adafruit/Adafruit_GPS #118: Possible logger example issue on UNO

caternuson, 2020-03-09:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=31&t=163050

Basic GPS and SD Card Info work.

Trying to run the logger example:
https://github.com/adafruit/Adafruit_GPS/tree/master/examples/shield_sdlog
and getting a compiler warning:
```
Low memory available, stability problems may occur.
```
and unexpected sketch behavior - serial output in setup() loops continuously.

caternuson, 2020-03-09:
As a work around, can revert to an older version of the library:
https://forums.adafruit.com/viewtopic.php?f=31&t=163050#p802610

drak7, 2020-03-24:
Looks like all of the improvements have added some weight to the library. When combined with another heavy library (SD) it uses quite a bit of RAM. Using the old version is one option, there are also some other purposefully small GPS libraries out there.

Should we look at trimming down or adding a lightweight option?

caternuson, 2022-07-18:
Maybe. But for now, closing this one. It was resolved by updating guides with call outs / warnings about UNO usage.
https://learn.adafruit.com/adafruit-ultimate-gps-logger-shield/soft-serial-connect
https://learn.adafruit.com/adafruit-ultimate-gps-logger-shield/sd-logging

closed 2022-07-18 by caternuson