adafruit/Adafruit_GPS #119: library not working, after last update library cant recover any data with the arduino mega and the v3 gps

jorgeneo560, 2020-03-18:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

ladyada, 2020-03-18:
please edit and complete the template

jorgeneo560, 2020-03-24:
the library is doing a bad job reading the serial buffer, it can't work at any other rate that is not in the example, i am using a arduino mega 2560 and compiler version 1.8.12

[sketch_mar16b.zip](https://github.com/adafruit/Adafruit_GPS/files/4372942/sketch_mar16b.zip)

drak7, 2020-03-24:
Can you be more specific about what's happening? A few things can cause issues when you want to do fast updates.

In your sketch you're requesting the RMC and GGA sentences along with antenna info at 10Hz. The GPS serial is at 9600bps and that amount of data is about 160 bytes, so you're requesting 1600 bytes per second but can only send 1200. If you reduce the amount of data to just the RMC sentence (use PMTK_SET_NMEA_OUTPUT_RMCONLY) you'll get better results there. You can also increase the GPS serial speed, check the PMTK_SET_BAUD commands.

ladyada, 2020-04-13:
closing for lack of response

closed 2020-04-13 by ladyada