adafruit/Adafruit_GPS #126: Parsing issue - latitude, others?

caternuson, 2020-07-21:
Related:
https://forums.adafruit.com/viewtopic.php?f=31&t=167318

Example sketch below parses a provided sentence, using the example data from here:
http://aprs.gids.nl/nmea/#gga
GGA latitude = "4124.8963" should parse out to 41 degrees and 24.8963 minutes, or 41.4149 degrees.

A spurious 1 is showing up, throwing off the computed value

```cpp
#include <Adafruit_GPS.h>

// Constructor for no communications, just data storage
Adafruit_GPS GPS;

// checksum calculator: https://nmeachecksum.eqth.net/
#define SENTENCE "$GPGGA,170834,4124.8963,N,08151.6838,W,1,05,1.5,280.2,M,-34.0,M,,*75"

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("GPS Parser Test");

  Serial.println("-------------------------------");

  if (GPS.parse(SENTENCE)) {
    Serial.print(GPS.latitude); Serial.print(" -> "); Serial.println(GPS.latitudeDegrees);
    Serial.print(GPS.longitude); Serial.print(" -> ") ; Serial.println(GPS.longitudeDegrees);
    // add any other parameters of interest
  } else {
    Serial.println("Parser Error.");
  }

  Serial.println("==============================");

}

void loop() {
  if (GPS.parse(SENTENCE)) {
    Serial.print(GPS.latitude); Serial.print(" -> "); Serial.println(GPS.latitudeDegrees);
    Serial.print(GPS.longitude); Serial.print(" -> ") ; Serial.println(GPS.longitudeDegrees);
    // add any other parameters of interest
  } else {
    Serial.println("Parser Error.");
  }
  delay(1000);
}
```

Running on UNO gives:
```
GPS Parser Test
-------------------------------
41241.89 -> -16.80
8151.68 -> -81.86
==============================
41241.89 -> -16.80
8151.68 -> -81.86
41241.89 -> -16.80
8151.68 -> -81.86
```

drak7, 2020-07-21:
This was fixed in 1.5.1, I'm not seeing any issues testing on an Uno. Is this still happening in the latest version?

caternuson, 2020-07-21:
Thanks. That seems to be it. I was still on 1.5.0. My bad. :facepalm: 

```
GPS Parser Test
-------------------------------
4124.90 -> 41.41
8151.68 -> -81.86
==============================
4124.90 -> 41.41
8151.68 -> -81.86
4124.90 -> 41.41
8151.68 -> -81.86
```

closed 2020-07-21 by caternuson

couillonnade, 2023-04-27:
The problem is there in 1.7.2

couillonnade, 2023-04-27:
OK, it was not clear in the sample but reading at the source code, there are two fields
latitude, and latitudeDegrees