adafruit/Adafruit_GPS #127: Milliseconds value appears to be incorrect

meltingSnowdrift, 2020-07-31:
**Arduino board:** Trinket M0
**Arduino IDE version:** 1.8.13
**Steps to reproduce:**
- Connect an Adafruit Mini GPS PA1010D module to an Adafruit Trinket M0 over I2C.
- Upload and run the GPS_I2C_Parsing demo sketch.

It appears that the milliseconds value provided by the library is incorrect.

After observing the output for a while under various settings, the milliseconds value output by the parser appears to be somehow dependent on time, probably on the seconds value. It tends to change, but possibly does not always change, when the seconds value changes.

Below is an example of output from the GPS_I2C_Parsing example sketch which demonstrates this. Note the difference between the milliseconds values provided by the parser and those present in the messages from the GPS module.

```
Time: 20:11:00.35552
Date: 31/7/2020
Fix: 1 quality: 1
Location: 4354.4805N, 7927.3735W
Speed (knots): 0.61
Angle: 9.15
Altitude: 300.20
Satellites: 12
$GNGGA,201101.000,4354.4806,N,07927.3734,W,1,12,0.88,299.9,M,-35.3,M,,*47

$GNRMC,201101.000,A,4354.4806,N,07927.3734,W,0.31,9.15,310720,,,A*6D

$GNGGA,201102.000,4354.4806,N,07927.3734,W,1,12,0.88,299.6,M,-35.3,M,,*4B

$GNRMC,201102.000,A,4354.4806,N,07927.3734,W,0.09,9.15,310720,,,A*65

Time: 20:11:02.37552
Date: 31/7/2020
Fix: 1 quality: 1
Location: 4354.4805N, 7927.3735W
Speed (knots): 0.09
Angle: 9.15
Altitude: 299.60
Satellites: 12
$GNGGA,201103.000,4354.4806,N,07927.3733,W,1,12,0.88,299.5,M,-35.3,M,,*4E

$GNRMC,201103.000,A,4354.4806,N,07927.3733,W,0.13,9.15,310720,,,A*68

$GNGGA,201104.000,4354.4806,N,07927.3733,W,1,12,0.88,299.6,M,-35.3,M,,*4A

$GNRMC,201104.000,A,4354.4806,N,07927.3733,W,0.46,9.15,310720,,,A*6F
```

foodini, 2020-10-19:
I've submitted a PR for this bug: https://github.com/adafruit/Adafruit_GPS/pull/129

drak7, 2020-10-27:
Fixed in #129

closed 2020-10-27 by drak7