adafruit/Adafruit_GPS #130: Possible library issue with GPS.fix not being released

RoboticSith, 2020-12-02:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  TeensyLC (current), also observed on Teensy4.0, Arduino Mega, Arduino Micro, & Arduino Pro Mini 5V

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

Summary: I am using the EN pin on the Ultimate GPS Breakout V3 to simulate a loss of fix so I can test various parts of my code and circuitry.  When I disable the module via this method after a fix has been established, the GPS.fix variable does not change to FALSE.

1) Connect GPS module to any of the above controllers.
2) Run parsing code for RMC & GGA sentences at 1Hz.
3) Wait for fix to be established, verify that GPS.fix is TRUE.
4) Connect the EN pin to GND, observe that the GPS module turns off yet GPS.fix remains TRUE.

[GPS_ExtParse.txt](https://github.com/adafruit/Adafruit_GPS/files/5631553/GPS_ExtParse.txt)

drak7, 2020-12-15:
There's no way for the library to know that you're turning off the GPS module.
If you want to test a loss of fix you can manually set fix to false or feed fake sentences to the library with the appropriate fix value.
You could also use the secondsSinceFix method to see when the last fix was and set GPS.fix to false after a certain threshold.

svdrummer, 2020-12-16:
You could use the tinygps++ library as this has a function to yest how old the nmea sentence is. If it is older than a few seconds, you can trap that in your code.

RoboticSith, 2020-12-18:
Thanks for the feedback.  I figured that the variable(s) behind the sentence contents would revert to their default/init values when the module was disabled since there wouldn't be any data behind them, I didn't think the previous sentence would stick around.

After posting this, I figured out that if I re-enabled the module briefly so it could send a handful of new sentences after having lost the fix (but not long enough to reacquire), everything in my code worked as expected.  I'll try one of the suggested solutions if I run into testing issues in the future once I incorporate the backup battery.

Thanks again.